SuperDARN/rst #375: [EHN] datdump

mts299, 2021-01-28:
# New Feature

**Name**: *datdump*

**function**:  *datdump.c* 

**bin/lib**: */datdump*

## Scope

### Reason

`dmapdump` displays the content of DMAP files; however, we don't have anything for dat files which is something we need for the Data archive set with FRDR and I would use it a lot. 

### Ability
Read in a dat file and display the contents in a terminal 

## Description

`datdump` will  print out the contents of a dat file similar to `dmapdump`:

Usage:
```
datdump 1999010101.t.dat 
```

### RST Checklist 

-  [x] Does this fit within RST's scope? 
-  [x] Is this a minor change?
-  [x] Have you created an RST project (github project) to reflect the process of developing it? 
-  [x] Does it follow GPL 3.0v License and has been copyrighted/authored to you? 

### Development help

- [x] scientific help with ensuring what you develop is scientifically correct/accurate

egthomas, 2021-01-29:
It doesn't print the contents of every field, but `test_raw` does most of what you're describing (for both `dat` and `rawacf` files).  Perhaps some kind of `-vb` option could be added to print out the remaining fields.

ksterne, 2021-01-29:
That and the `cmp_raw` sound like a helpful start.  I think we've captured they desire here of having some type of output that can be captured.  I'll take a look as these when I can, but I might not get to it quickly.

mts299, 2021-01-29:
Sorry, I forgot to mention I am working on this new feature, I was just giving a heads up. 
Thank you. 

Not looking for a solution.

ecbland, 2021-01-29:
Just to clarify what's going on here, since I realise it's come out of the blue...

We added some issue templates on github to help us share useful information when creating a new issue. There are several templates designed for different purposes. One of these is to notify other RST developers that a new feature is in the works. This allows any suggestions or potential problems to be identified before the PR stage. That's the template that @mts299 used here, but I can see now that the purpose wasn't clear.

So, @mts299 is actively working on this new feature, and suggestions/feedback are most welcome here.

Sorry for any misunderstandings about this: I've put this on the agenda for our next meeting so we can have a brief discussion about it. And we can improve the [templates](https://github.com/SuperDARN/rst/tree/master/.github/ISSUE_TEMPLATE) as needed with a PR :smile:

ecbland, 2021-02-01:
@mts299 This issue has caused a lot of confusion since the PR appeared only 1 day after the issue was opened. Perhaps we can close this and have the discussion over at #376?

closed 2021-02-01 by mts299