SuperDARN/rst #378: [BUG] time_plot not plotting South pole data with -geo -mag option

mts299, 2021-02-03:
# BUG 

While working on #348 I noticed the south pole data is not plotted when you use `-mag` or `-geo` options with `time_plot`

## Priority

- [x] Minor - typo, incorrect naming, fails on specific uncommon situation
- [ ] Major - misrepresentation of data, failure in a common situation 
- [ ] Urgent - failure in usage, large misrepresentation

## Example of the bug 

```
time_plot -png -a  20120505.0000.16.san.fitacf > 20120505.0000.16.san.dev.png
```

[image]

```
time_plot -png -a -mag -latmin 72  20120505.0000.16.san.fitacf > 20120505.0000.16.san.dev.mag.png
```

[image]

### Attempts

Have not done anything to try and fix it. If someone wants to fix it please use assignee on the issue so others know you are going to fix it. 

## Potential Bug Location 
 
time_plot.c file

mts299, 2021-02-03:
Looking at the code, not sure where the fix would need to be, if someone else can take a crack at it that would be much appreciated.

mtwalach, 2021-02-09:
Hi @mts299 - Just from looking at the figure, Imaybe the problem is the fact that the latitude should be negative since it's in the southern hemisphere? Just a guess though as I have not used this particular code myself...

mts299, 2021-02-09:
@mtwalach thanks! yea I saw @egthomas used `abs` for the grid plotting bug with the south so I am hoping he might pick up this issue as well. 

The code is a bit confusing to me so it will take a bit more time for me.

closed 2022-08-24 by egthomas