SuperDARN/rst #389: Filtering non-Gaussian noise/interference

pasha-ponomarenko, 2021-02-14:
# New Feature

**Name**: Filtering non-Gaussian noise/interference

**function**:  salt_and_pepper_filter.c

**bin/lib**: */directory*

## Scope

*Create a check list using markdowns [checklist syntax](https://help.github.com/en/github/managing-your-work-on-github/about-task-lists) on the scope of the feature.*

Thisi feature will effectively remove excessive salt-and-pepper noise and vertical "stripes" caused by non-Gaussian noise/interverence sources from FITACF3 output. 

*What will the feature not do?* 

It will not lalter the fitted parameters, just remove the erroneous/contaminated data from fruther analysis.

## Description

*Please provide a description of the new feature, and whether it will improve/solve a problem.*
[Filtering non-Gaussian interference.pptx](https://github.com/SuperDARN/rst/files/5978191/Filtering.non-Gaussian.interference.pptx)

### RST Checklist 

-  [ yes ] Does this fit within RST's scope? 
-  [ yes ] Is this a minor change?
-  [ may be ] Is this a major change?
-  [ yes ] Do you need help developing it? 
-  [ no ] Have you created a RST project (github project) to reflect the process of developing it? 
-  [ don't know ] Does it follow GPL 3.0v License and has been copyrighted/authored to you? 

### Developement help

*RST does not have a dedicated development team for full feature development. 
However, if you are able to provide some time and help, the community will try to aid you the best way they can in making your feature possible.*

*If you need help please indicate what expertise you are looking for:*
- [ ] developer help with design and integration of code
- [ ] scientific help with ensuring what you develop is scientifically correct/accurate
- [ ] project management to help organize what steps need to be taken to make this possible 

*Please also provide any other information you may need help with (not including testing/reviewers of code)*

## User Interface 

*Please provide **pseudocode** on how you see your new feature interacting with the code or user-interface*

## Extra Notes

*Please provide any other details about this feature*
- *publications*
- *GitHub link of code in another language*
- *potential solutions to current/future problems*

ecbland, 2021-02-17:
Thanks for this @pasha-ponomarenko :100: 

It looks like slide 4 of your powerpoint presentation contains all the info I need to test this...is that right?

pasha-ponomarenko, 2021-02-17:
G'day, Emma!
Yep, this is essentially it I believe. Besides testing the filter itself we may also look at how we can trigger it automatically, e.g., based on the percentage of the "salt-and-pepper" returns.

ecbland, 2021-02-18:
@pasha-ponomarenko The median filtering works well (no surprises there). One thing to watch out for is frequency switching (e.g. `twofsound`), since the backscatter location from one scan to the next can be very different. Increasing the size of the filter helps here, as shown in this example from RKN: [20180901_rkn.pdf](https://github.com/SuperDARN/rst/files/6005032/20180901_rkn.pdf)

pasha-ponomarenko, 2021-02-18:
Thanks a lot, @ecbland !
I think that it would be more appropriate from Physics point of view to split the frequency channels before filtering rather than to increase the window size. I guess these can be done by using prm.channel field.
Is it hard to implement?

ecbland, 2021-03-02:
@pasha-ponomarenko I've started working on implementing this code in C. It will not be a problem to split the data into frequency channels (we also have to split it by beam numbers). The code will do the following:
1. Split data by channel and beam number
2. Do the median filtering for each beam/channel separately
3. Write a new fitacf file with the salt & pepper noise removed, but otherwise identical to the original file

A filter width of 3 seems to work best based on the testing I have done. We can also allow the user to choose the filter width with a command line option. 

> we may also look at how we can trigger it automatically, e.g., based on the percentage of the "salt-and-pepper" returns.

I have not looked into this at all yet, and I do worry about introducing more ad hoc criteria into the analysis software. Maybe we can come back to this later if necessary?

I'm juggling a few things at the moment but I hope to post some test code soon.

pasha-ponomarenko, 2021-03-02:
@ecbland , thanks a lot for pushing this issue forward! I think we need to flag somehow -- a new field in the prm structure? -- indicating that filtering has been applied. 
I feel more conservative with respect to the optional filter size as it  would introduce yet another source of apparent data incompatibility.

pasha-ponomarenko, 2021-03-02:
Oh, and I agree with your view on the automatic triggering. 
:-)

closed 2021-05-17 by mts299