SuperDARN/rst #392: trim_raw producing empty output files

ecbland, 2021-03-01:
# BUG 

`trim_raw` produces an empty output file when trying to trim a `rawacf` file. 
`dat` files are not affected.

## Priority

- [ ] Minor - typo, incorrect naming, fails on specific uncommon situation
- [ ] Major - misrepresentation of data, failure in a common situation 
- [X] Urgent - failure in usage, large misrepresentation

## Example of the bug
```
trim_raw -vb -st 22:30 -et 23:00 20140214.2201.00.inv.rawacf > out.rawacf

# Check the size of the output file:
ls -lh out.rawacf
-rwxrwxrwx 1 emma emma   0 Mar  1 10:31 out.rawacf
```

### Attempts

*What have you tried already to fix it?*
Nothing

*What have you tried to get around it?*
Use RST4.4.1 (i.e. the bug seems to have been introduced in RST4.5)

### Data Location

Any rawacf file from Globus/BAS

## Extra Notes

*Please provide any other details about this bug*
- *Operating System* `Linux Mint 20 Cinnamon`
- *Compiler version* `gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0`

egthomas, 2021-03-01:
This worked correctly as of 8b7fe8cbe6b89dd40ea766a62f9cf5003f4801ba - perhaps subsequent changes to the file introduced an error.

ecbland, 2021-03-01:
I think I found it - working on it now

PR #395 (2021-03-02): Fix bug in trim_raw
This PR fixes #392. This was a silly mistake that crept in when adding comments to `trim_raw.c` in #320, where the line `RawFwrite(stdout,prm,raw);` was accidentally removed after the code itself had been tested. That's the line that writes the output file :stuck_out_tongue_winking_eye: 

To test:
```
trim_raw -st 08:00 -ex 00:30 20140220.0802.00.bks.rawacf > out.rawacf

#check output file size
ls -lh out.rawacf
```
On this branch the output file will contain the trimmed data; on develop the file will be empty
closed 2021-03-02 by mts299