SuperDARN/rst #393: grid_plot  -x doesn't stay open

mts299, 2021-03-01:
# BUG 

`grid_plot -x` does not keep the display window open when displaying the data. 

## Priority

- [x] Minor - typo, incorrect naming, fails on specific uncommon situation
- [ ] Major - misrepresentation of data, failure in a common situation 
- [ ] Urgent - failure in usage, large misrepresentation

## Example of the bug 

`grid_plot -x -raw -st 00:02 -et 00:08 -time -coast -xkey rainbow.key -fcoast 20100305.pgr.grid`

While trying to test pyDARN `plot_grid` method, I was using 'grid_plot' as verification and wanted to use `-x` option to just display instead of making an actual plot file. However, it opens then closes the window even when it hits the end time. 

### Attempts

I just tried various time options to see if it would get fixed. 

### Data Location

I am only testing with a single radar grid file so just need to make `make_grid` on a fitacf file. 

## Potential Bug Location 

In the grid_plot.c code where I think `-x` is getting closed to prevent multiple windows open. I assume the best solution is the last timestamp should not close the window. So determining when that is and then preventing the window to close. 

## Potential Solution(s)

A nice elegant solution would actually give the user control to close each window or move to the next instead of it closing them as the time stamp moves on.

mts299, 2021-03-01:
@ecbland just told me `-delay` is what I need to use to fix the problem. 

Would it be good to maybe have -delay 0 as the default so that this `-x` option doesn't seem pointless without `-delay 0`?

ecbland, 2021-03-01:
Yes, adding `-delay 0` lets the user click through the plots at their leisure. I don't feel strongly about what the default behaviour should be. Any other opinions here?

Note that `map_plot` is similar to `grid_plot`, so both should behave the same.

closed 2021-06-21 by ecbland