SuperDARN/rst #410: ELV_HIGH lower than ELV in many FitACF v2.5 cases

alexchartier, 2021-04-19:
# BUG 

The "ELV_HIGH" field in fitACF v2.5 files frequently contains values lower than the corresponding "ELV" values. 
This is contrary to docs, which indicate ELV_HIGH contains the highest estimate of elevation. 

## Priority

- [ ] Major - misrepresentation of data, failure in a common situation 

## Example of the bug 

See, e.g. 20140423.cvw.fit, 20140423.kod.d.fit (created using fitACF from all BAS rawACFs on 23 April). 

In case of 20140423.cvw.fit (read using Pydarn):

elv
['32.40', '12.04', '12.94', '18.51', '35.22', '10.87', '12.63', '34.65', '20.58', '21.48', '17.18', '22.70', '20.33']
elv_high
['32.08', '9.59', '9.45', '15.82', '34.88', '6.50', '11.06', '34.09', '19.05', '19.80', '14.95', '19.93', '17.54']

## Potential Bug Location 

cross-correlation code

## Potential Solution(s)

fix the elevation angle algorithm

## Extra Notes

ecbland, 2021-04-19:
Thanks for the bug report @alexchartier! I also stumbled on this issue a few years ago, and I remember discussing it at a SuperDARN workshop (probably with @pasha-ponomarenko), but we never took any action.

The bug affects radars that have their interferometer array located **behind** the main array. RST does not check this when writing the `elv_low` and `elv_high` fields, so the values get swapped. This is probably straightforward to fix. 

**A question for you:** Are you planning to use `elv_low` and `elv_high` values for research, or did you just notice this bug by chance? I'm asking because the upper & lower elevation estimates were replaced with a single least-square error estimate in FITACF3.0 (see [release notes](https://superdarn.github.io/dawg/files/sup_material/FITACF3_release_notes.pdf)). If people are using them for science then we need to know!

alexchartier, 2021-04-19:
Emma,

Yes, for science. I think the bug should be fixed, especially as fitACF 2.5 is in wider use than FitACF 3.

Alex

pasha-ponomarenko, 2021-04-20:
@ecbland , I cannot recall this bug... :-( 
Anyways, if we want to fix it then it probably makes sense to introduce a sign multiplyer in the same way as it is introduced in elevation.c: 

>    if (prm->interfer[1] > 0.0) /* interferometer in front of main antenna */
>       phi_sign= 1.0;
>     else {                           /* interferometer behind main antenna */
>       phi_sign= -1.0;  

and apply it to phase error in the respective expresions for _elv_high_ and _elv_low_ in _do_fit.c_.  

@alexchartier , on a side note, the phase error estimates in FITACF2.5 are based on somewhat intuitive approach so those should be taken with a grain of salt.

ecbland, 2021-04-20:
@alexchartier I've fixed the bug in PR #411 -- please let us know if you're able to help with testing it.

ecbland, 2021-04-22:
Closing this now that #411 has been merged

PR #411 (2021-04-22): Fix elevation angle error bounds in FITACF2.5
This PR fixes #410 reported by @alexchartier, where the elevation angle error bounds (`elv_low` and `elv_high`) are flipped for some radars. The problem occurs for radars with the interferometer array behind the main array. 

I fixed this problem for the default (Shepherd) elevation angle calculation, and also for the `-old_elev` option. For consistency I corrected the function call to `elev_goose()`, but that radar has a front inteferometer so it doesn't change the behaviour. 

To test (with and without `-old_elev`):
```
make_fit [-old_elev] 20100 [... cut]
closed 2021-04-22 by ecbland