SuperDARN/rst #415: [EHN] map_addimf removing the need for ISTP environment variable

mts299, 2021-04-23:
# New Feature

Allow flexibility to input directory name instead of setting `ISTP`. 

PLEASE NOTE: I am not suggesting deprecating `ISTP` just adding on a new user-friendly option. 

## Scope

Side note: out of scope but would be nice it grab it from the websites :p 

- [ ] `-ace` `wind` allows a user to input a directory to grab data from instead of using `ISTP`
- [ ] ISTP still works if nothing is set for a directory
- [ ] error is returned if directory or `ISTP` directory is empty 

## Description

The new feature will check and allow the user to specify a directory to find ace or wind data instead of setting a `ISTP` variable which may be a bit complicated for people who don't know where to find the `.bashrc` file in RST
 
### RST Checklist 

-  [x] Does this fit within RST's scope? 
-  [x] Is this a minor change?
-  [ ] Is this a major change?
-  [ ] Have you created a RST project (github project) to reflect the process of developing it? 
-  [x] Does it follow GPL 3.0v License and has been copyrighted/authored to you? - its a modification

## User Interface 

``` bash
map_addimf -ace /data/ace/2018/ 20181001.hmb.map > 20181001.imf.map
```

ecbland, 2021-05-10:
> User Interface   `map_addimf -ace /data/ace/2018/ 20181001.hmb.map > 20181001.imf.map`

Just a quick clarification question: with this change, `map_addimf` will accept 2 input arguments if provided? 
And will you keep the `-p` option?

mts299, 2021-05-11:
typically flexibility for the user. If `-p` still enforces a structure for the user then this is less user friendly

mts299, 2021-05-18:
I will come back to this a bit later still too busy

closed 2021-05-18 by mts299