SuperDARN/rst #426: Next minor release: RST4.6

ecbland, 2021-06-01:
We are now due for a new minor release of the RST :rocket: 

Several new features have been added since the previous release in mid-January, including:

- `fit_speck_removal`, which removes noise/interference from FITACF files
- Update FITACF 3.0 to use the Shepherd (2017) elevation angle calculations (`elevation_v2`)
- New `datdump` library for directly reading old-format `dat` files. Kathryn has requested this feature to support the publication of a SuperDARN dataset on FRDR
- Updated documentation

There have also been several bug fixes in the core analysis code. 

There are two open issues that I would like to address for this release:
- #345 -- I've been in touch with the PI about updating the hardware files to fix this bug
- #399 -- this bug was introduced in the previous release
- Anything else?

pasha-ponomarenko, 2021-06-01:
@ecbland , while it is tempting to include a quick fix for the [BOREALIS low noise issue](https://github.com/SuperDARN/rst/issues/419), it might be prudent to elaborate a more general solution that would not affect the old data.

ecbland, 2021-06-11:
An update on the next release: 

- [X] #345 -- will be fixed with an update to the hardware file. We are just waiting for the PI to confirm the change with their engineer. We can update the hardware file on the release branch when it's ready
- [X] #399 -- has been fixed
- [ ] #433 -- looks easy to test and would be a very helpful addition, so let's include that in this release
- [ ] #430 -- always good to fix bugs, so I will test this ASAP

I think everything else can wait for a future release. Hopefully this means we can create the release branch early next week.

ksterne, 2021-06-11:
Is it worth it to add in #397?  It seems like that is a good feature to add in and looks ready to be tested.

ecbland, 2021-06-18:
Ok, I've merged the updates to MLTv2 and `fov_plot` that @egthomas provided. 

Thanks also to @mts299 for looking into #433. This turned out to be a more difficult change than anticipated due to memory allocation issues, so let's wait until a later release to implement an alternative solution (e.g. a new field in the dmap files)

@mts299 offered to make the release branch, and I think we are ready to do that now :rocket:

ksterne, 2021-06-18:
Is it worth it to add in #397?  It seems like that's a good to add in.

pasha-ponomarenko, 2021-06-18:
@ecbland , I am contemplating a quick fix for https://github.com/SuperDARN/rst/pull/420 as it is a rather annoying bug. The idea is to add a second condition that the "search" noise not equal to zero. Then the old data will remain unchanged while Borealis data will be corrected. A more elaborate resolution will be left for the next release.
What do you think?

ecbland, 2021-06-18:
@pasha-ponomarenko Are you able to make this change today?

pasha-ponomarenko, 2021-06-18:
Will try but right now I am trying to fix my Linux box -- it is stuck at 640x480 resolution

closed 2021-08-10 by ecbland