SuperDARN/rst #443: "map_addimf" does not read correctly TXT files when "-old" option is used working with old map files

gighen74, 2021-07-14:
I accidentally found this bug/inconsistency, which seems to be present in Release 4.6 and also in the previous one, v. 4.4.1
When one tries to add interplanetary magnetic field, solar wind and Kp information to a map, passing a TXT file to "map_addimf", the Kp field is not read if and old map file is processed. Instead, if a cnvmap is processed, the TXT is correctly read with all its fields. 
Full details below in the attached file
[command_list.txt](https://github.com/SuperDARN/rst/files/6815090/command_list.txt)

This small bug is probably present since the introduction of TS18_Kp model in RST.

Thanks

Gigi Coco

egthomas, 2021-07-14:
Hi @gighen74, yes you are correct that Kp information is not properly stored in old-format map files. This was because we did not want to break any legacy software by adding additional columns or fields to the old-format map files.  You might notice that the solar wind Vx information is stored in the Bx field of the old-format map files but unfortunately there did not seem to be any other unused fields we could co-opt to store Kp information.

One workaround could be to keep your `grdmap` as-is, drop the `-old` flag from the `combine_grid` and Map Potential processing routines, and then at the end convert the `cnvmap` format map files back to the older plain-text map file format with `cnvmaptomap`.  I know it's not ideal but it should at least ensure the correct background statistical model is used for the fitting.

gighen74, 2021-07-14:
Dear Evan,

Ok, I understand. In fact I already apply the workaround you suggest, and
this works well enough.
Perhaps it would be useful to add a note somewhere in the documentation for
this issue, if not even in the --help description of the command
map_addimf: beside the -if option one could add that the option does not
allow to grab Kp from the IMF file if -old is used at the same time...

Thanks for your prompt answer

Cheers

Gigi

egthomas, 2021-07-14:
Hi Gigi, my apologies for not reading your original message as closely as I should before responding.  Your suggestion to update the documentation sounds like a good one - perhaps @ecbland, @mts299, or @ksterne know how to add this information to the "GRID to MAP" documentation on the new website:

https://radar-software-toolkit-rst.readthedocs.io/en/latest/user_guide/map_grid.html

ksterne, 2021-07-14:
I can certainly take on updating the documentation here.  I think that's something we could put directly in the release branch unless there's an objection otherwise.

ksterne, 2021-07-14:
@egthomas, maybe refresh my memory here but I believe the section on the `map_grid.html` should be updated as there are additional fields after `second bx by bz`.  It looks like the additional ones should be `Vx Kp`, is this correct?  Is the Kp only needed when using the TS18_Kp model?  I feel like we had this down in documentation somewhere, but I can't seem to find it now.  Apologies for that!

ksterne, 2021-07-14:
@gighen74, for now, I think 1da6108 and 6caa208 that I've added to the `release/rst-4.6` branch should address the behavior of the `map_addimf` binary when using the `-if` and `-old` options together.  Let me know if there is something that I've missed.

gighen74, 2021-07-14:
Hi Kevin,

I think the notes you introduced in the documentation are clear enough.
Thanks

Cheers

Gigi

ksterne, 2021-07-17:
I've added in the `Vx Kp` documentation on the map_addimf with 56f729b, so I think we're good to clear/reoslve this issue, yes?

gighen74, 2021-07-17:
For me It's ok. Should I do it myself, as I'm the one who opened the issue?

ecbland, 2021-07-20:
@gighen74 Yes it's fine for you to close issues that you have opened. I'll close it this time since I'm here anyway :smile:

closed 2021-07-20 by ecbland