SuperDARN/rst #450: [BUG] failure to compile with intel compiler

mts299, 2021-08-13:
# BUG

Trying to upgrade RST on our high-performance computer that is shifting to using newer netcdf libraries that uses intel compiler. 
RST was unable to compile with the intel compiler (see below on error message).  
## Priority

- [ ] Minor - typo, incorrect naming, fails in a specific uncommon situation
- [ ] Major - misrepresentation of data, failure in a common situation
- [x] Urgent - failure in usage, large misrepresentation of data
However, I don't think this needs to be dealt with ASAP because this is one specific compiler that I later got working with gcc compiler and it seems most users may not use this setup. 

## RST version

- [x] Yes. RST version: 4.6
- [ ] No. Branch containing the bug (e.g. develop): ______

## Example of the bug

```bash 
make.version /project/6008057/rst/codebase/superdarn/src.lib/tk/elevation.1.0/src/..
make.hdr /project/6008057/rst/codebase/superdarn/src.lib/tk/elevation.1.0/src/../include /home/mschmidt/projects/rpp-kam136/rst/include/superdarn 
/project/6008057/rst/codebase/superdarn/src.lib/tk/elevation.1.0/src/../include/elevation.h
'/project/6008057/rst/codebase/superdarn/src.lib/tk/elevation.1.0/src/../include/elevation.h' -> '/home/mschmidt/projects/rpp-kam136/rst/include/superdarn/elevation.h'
cc -fPIC -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX -I/home/mschmidt/projects/rpp-kam136/rst/include/base -I/home/mschmidt/projects/rpp-kam136/rst/include/superdarn -c elev_goose.c    
mkdir -p /home/mschmidt/projects/rpp-kam136/rst/lib
ar -r /home/mschmidt/projects/rpp-kam136/rst/lib/libelevation.1.0.a elevation.o elevation_v2.o elev_goose.o
ld -shared -lm -lz  -o /home/mschmidt/projects/rpp-kam136/rst/lib/libelevation.1.0.so \
              elevation.o elevation_v2.o elev_goose.o -Bstatic -L/home/mschmidt/projects/rpp-kam136/rst/lib  
/cvmfs/soft.computecanada.ca/gentoo/2020/usr/x86_64-pc-linux-gnu/binutils-bin/2.33.1/ld: elevation_v2.o:(.bss+0x0): multiple definition of `FitPrm'; elevation.o:(.bss+0x0): first defined here
/cvmfs/soft.computecanada.ca/gentoo/2020/usr/x86_64-pc-linux-gnu/binutils-bin/2.33.1/ld: elev_goose.o:(.bss+0x0): multiple definition of `FitPrm'; elevation.o:(.bss+0x0): first defined here
make: *** [/home/mschmidt/projects/rpp-kam136/rst/build/make/makelib.linux:25: elevation] Error 1
```
Happened on: netcdf-4.8.0 with intel 2021
Essentially switching to netcdf/4.6.1 and gcc-7.3.0 it now compiles properly. 

### Attempts

I tried a couple of things in the elevation library include guards (`#ifndef`) and other linking methods.

## Potential Bug Location

It may be in elevation.c code or it might be some changed to the FITACF 3.0 using elevation library now.

ksterne, 2021-08-13:
Yikes!  Any advice on how to replicate the environment you're trying to set this up on for a high-power computer (HPC)?

mts299, 2021-08-13:
I don't know except using intel compiler. I will try to reproduce it on my ubuntu laptop.

closed 2022-03-15 by ecbland