SuperDARN/rst #465: Couldn't find png.h during make.code

JWiker, 2021-09-26:
When running `make.code`, I get the following error:

`cc -fPIC -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX -I/project/superdarn/software/rst_latest/include/base -c png.c    
png.c:32:17: fatal error: png.h: No such file or directory
 #include <png.h>
                 ^
compilation terminated.
make: *** [png.o] Error 1
Compilation Aborted.`

Running `find $RSTPATH -name "png.h"` returns no results.

Any help would be appreciated!

ecbland, 2021-09-27:
@JWiker, have you checked that you have installed `libpng-dev`?

https://radar-software-toolkit-rst.readthedocs.io/en/latest/user_guide/linux_install/#library-requirements

If that's not the problem, please let us know which operating system and `gcc --version` you're using so that we can try to help further.

ecbland, 2021-11-10:
@JWiker I haven't heard back from you, so I will close this issue now. Please re-open it if you still need assistance.

JWiker, 2021-11-10:
@ecbland Sorry about that, I was waiting for some internal feedback on this but just realized it never came.

I confirmed that libpng is installed, but we have `libpng12` instead of `libpng-dev`. I found `png.h` at `/usr/include/libpng12/png.h` - do I need to move it to a specific location within $RSTPATH?

JWiker, 2021-11-10:
Ok I think that problem is solved, but I've run into a new one that seems to be caused by using C90 instead of C99. 

```
In file included from rcdf.c:36:0:
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1918:8: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
   long long tt2000, double *year, double *month, double *day, ...
        ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1643:27: note: in definition of macro ‘PROTOARGs’
 #  define PROTOARGs(args) args
                           ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1920:21: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
 VISIBLE_PREFIX long long CDF_TT2000_from_UTC_parts PROTOARGs((
                     ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1924:8: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
   long long time
        ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1643:27: note: in definition of macro ‘PROTOARGs’
 #  define PROTOARGs(args) args
                           ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1926:21: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
 VISIBLE_PREFIX long long CDF_TT2000_from_UTC_EPOCH PROTOARGs((
                     ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1930:8: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
   long long time, double *epoch16
        ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1643:27: note: in definition of macro ‘PROTOARGs’
 #  define PROTOARGs(args) args
                           ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1932:21: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
 VISIBLE_PREFIX long long CDF_TT2000_from_UTC_EPOCH16 PROTOARGs((
                     ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1936:8: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
   long long time, char *string, ...
        ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1643:27: note: in definition of macro ‘PROTOARGs’
 #  define PROTOARGs(args) args
                           ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1938:21: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
 VISIBLE_PREFIX long long CDF_TT2000_from_UTC_string PROTOARGs((
                     ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1941:21: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
 VISIBLE_PREFIX long long computeTT2000withBasedLeapDay PROTOARGs((
                     ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1946:8: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
   long long tt2000, int yymmdd, 
        ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1643:27: note: in definition of macro ‘PROTOARGs’
 #  define PROTOARGs(args) args
                           ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1950:21: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
 VISIBLE_PREFIX long long parseTT2000withBasedLeapDay PROTOARGs((
                     ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1954:8: warning: ISO C90 does not support ‘long long’ [-Wlong-long]
   long long tt2000,
        ^
/project/superdarn/software/rst_latest/include/analysis/cdf.h:1643:27: note: in definition of macro ‘PROTOARGs’
 #  define PROTOARGs(args) args
                           ^
rcdf.c: In function ‘RCDFReadZ’:
rcdf.c:219:5: error: ‘for’ loop initial declarations are only allowed in C99 mode
     for (int i=0;i<cnt;i++) 
     ^
rcdf.c:219:5: note: use option -std=c99 or -std=gnu99 to compile your code
rcdf.c:221:37: warning: pointer of type ‘void *’ used in arithmetic [-Wpedantic]
         memcpy(buffer[i],(void*)data+offset[i],size[i]);  
                                     ^
rcdf.c: In function ‘RCDFReadR’:
rcdf.c:332:37: warning: pointer of type ‘void *’ used in arithmetic [-Wpedantic]
         memcpy(buffer[i],(void*)data+offset[i],size[i]);  
                                     ^
In file included from rcdf.c:36:0:
rcdf.c: At top level:
/project/superdarn/software/rst_latest/include/analysis/cdf.h:80:16: warning: ‘TT2000NULL’ defined but not used [-Wunused-variable]
 static double *TT2000NULL = 0;
                ^
make: *** [rcdf.o] Error 1
Compilation Aborted.
```
gcc --version gives mt: `gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-39)`

ecbland, 2021-11-12:
@mts299 Could you please help @JWiker with the compilation question? Thank you :)

JWiker, 2021-11-16:
@mts299 Is there any harm in adding `-std=c99` to the makefiles so that others trying to build RST on older machines don't run into that issue?

mts299, 2022-01-11:
Hi @JWiker sorry I didn't notice this issue, there is no issue but we updated the code to not have to run on c99 so weird you have to specify

JWiker, 2022-01-11:
That is weird - Rob Barnes and I weren't able to build the latest RST unless we explicitly specified c99. In any case, it's working now.

mts299, 2022-01-11:
I will close this issue then and maybe just keep your issue in mind.

closed 2022-01-11 by mts299