SuperDARN/rst #467: Issue using fit_speck_removal

JWiker, 2021-11-16:
After pulling the latest RST/master and building fit_speck_removal, it does not seem to be working as expected. Did the build somehow silently fail, or am I just not using it correctly?

I ran it as follows, attempting to despeckle 20140424.kod.c.v3.0.fit, and it spits out all of this for a few minutes, then doesn't seem to actually change the file:

```
fit_speck_removal 20140424.kod.c.v3.0.fit
?3(radar.revision.majorradar.revision.minororigin.codeorigin.time	Tue Nov 16 20:10:32 2021origin.command	fit_speck_removal 20140424.kod.c.v3.0.despeckled.fitcpA?stidtime.yr?time.motime.dytime.hrtime.mttime.sctime.u<
                                     txpow(#naveattenlagfr?smsepdercodstat.agcstat.lopwrnoise.search??Cnoise.meanchannelbmnum	bmazm?p
                                                                                                                                       Bscanoffsetrxrisedintt.scintt.ustxpl?mpinc?mppumplgsmplgexsifmodenrang?frang?rsepxcftfreq?4mxpwr@lvmax Ncombf	pcodecamp_bm9_15kmfitacf.revision.majorfitacf.revision.minornoise.sky般Bnoise.lag0noise.velptaltab*++*+++pwr0?H?H"??fȾH?Q
??H?H?H?}҅?]?2?H?H?H???@?Q=??H?H?H?H?H?H?o??H?H??91?H?H?H?H?HM??H?C???H?H?u$F?eݧ???x??{??r?@?8)@C."??a???t???O?B?<???H?H??N??1ci?H????eT@$@H?_a??H???(?H?H?Ha??H?F,??\??od%?A?¾H7??2`?w{5???>???w???@?:@&&?@???@w
                         @??6@???@?A????[??@???ƻ??õ?@??@????3aA@N?A?????@?2'@k}?@1?T@??V?ˀ??߈@S???b???@??b>.y?@n?@?l@D?0??!?>*@Bf?@?y9@???????
}?
??????H???*?D$??R@????" @?؊@?B&??z?H?i???H??????X??8?=?\?????m?H???|???????H?N???????l?H?>?????p???H?H??4??H?H?H?B????@&?H???u$F?+???H??3?HB??U?????(??F??O???Ȇ??[??Y?%?H?8?C???q???L?H?{??1
```

ksterne, 2021-11-16:
I think this looks like normal behavior.  As with the `make_fit` routine, I think you should pipe the output into a file.  So something more like:
`fit_speck_removal 20140424.kod.c.v3.0.fit > 20140424.kod.c.v.3.0.fit.despeckle`
If you can point to documentation that says otherwise on how to use the routine, that would be helpful for us to update it.

JWiker, 2021-11-16:
Ah ok, I was thinking it would update/replace the speckled file with a despeckled version, and would only print something if `-vb` was used. Thanks @ksterne!

I think it'd be helpful to add that syntax to the help info - right now, that `fit_speck_removal [-vb] fitacfname` line is what made me think you don't need to specify an output file:
```
[superdarn@tuvalu 04]$ fit_speck_removal --help
fit_speck_removal - Removes salt and pepper noise from a fitacf file.
fit_speck_removal --help
fit_speck_removal [-vb] fitacfname
--help     print the help message and exit.
--version  print the RST version number and exit.
-vb        verbose. Log information to the console.
fitacfname filename of the input fitacf-format file.
```

ecbland, 2022-01-11:
> I think it'd be helpful to add that syntax to the help info - right now, that fit_speck_removal [-vb] fitacfname line is what made me think you don't need to specify an output file:

I see where this confusion comes from, and in fact it's not unique to `fit_speck_removal.` The command line help is formatted the same for most RST binaries (i.e. it doesn't specify that the user should pipe the output to a file)

The [readthedocs documentation](https://radar-software-toolkit-rst.readthedocs.io/en/latest/) provides tutorial-style instructions on how to use the RST binaries. I'm already planning to add some documentation on `fit_speck_removal` to that, so I will close this issue now.

closed 2022-01-11 by ecbland