SuperDARN/rst #475: CFLAG required for compilation

aburrell, 2022-02-28:
# BUG

When compiling my code on:

LSB Version:	:base-4.0-amd64:base-4.0-noarch:core-4.0-amd64:core-4.0-noarch:graphics-4.0-amd64:graphics-4.0-noarch:printing-4.0-amd64:printing-4.0-noarch
Distributor ID:	CentOS
Description:	CentOS release 6.10 (Final)
Release:	6.10
Codename:	Final

with:

cc (GCC) 4.4.7 20120313 (Red Hat 4.4.7-23)
Copyright (C) 2010 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I get the following error:

```
================================================================================
Compiling Library:/home/aburrell/Git/rst/codebase/analysis/src.lib/cdf/rcdf.1.5
================================================================================
make clean
rm -f *.o
make
make.version /home/aburrell/Git/rst/codebase/analysis/src.lib/cdf/rcdf.1.5/src/..
make.hdr /home/aburrell/Git/rst/codebase/analysis/src.lib/cdf/rcdf.1.5/src/../include /home/aburrell/Git/rst/include/analysis 
/home/aburrell/Git/rst/codebase/analysis/src.lib/cdf/rcdf.1.5/src/../include/rcdf.h
`/home/aburrell/Git/rst/codebase/analysis/src.lib/cdf/rcdf.1.5/src/../include/rcdf.h' -> `/home/aburrell/Git/rst/include/analysis/rcdf.h'
cc -fPIC -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX -I/home/aburrell/Git/rst/include/analysis -I/home/aburrell/bin/cdf38_0-dist/include -c rcdf.c    
In file included from rcdf.c:36:
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2000: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2003: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2006: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2009: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2012: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2015: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2018: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2021: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2024: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2027: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2030: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2034: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2039: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2042: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2047: warning: ISO C90 does not support ‘long long’
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:2052: warning: ISO C90 does not support ‘long long’
rcdf.c: In function ‘RCDFReadZ’:
rcdf.c:219: error: ‘for’ loop initial declarations are only allowed in C99 mode
rcdf.c:219: note: use option -std=c99 or -std=gnu99 to compile your code
rcdf.c:221: warning: pointer of type ‘void *’ used in arithmetic
rcdf.c: In function ‘RCDFReadR’:
rcdf.c:332: warning: pointer of type ‘void *’ used in arithmetic
make: *** [rcdf.o] Error 1
Compilation Aborted.
```

I tried the suggested fix out in the problem directory, and it appeared to work:

```
cd $RSTPATH/codebase/analysis/src.lib/cdf/rcdf.1.5
make clean
cc -fPIC -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX -I/home/aburrell/Git/rst/include/analysis -I/home/aburrell/bin/cdf38_0-dist/include -c rcdf.c -std=gnu99
```

The output from this call is:

rcdf.c: In function ‘RCDFReadZ’:
rcdf.c:221: warning: pointer of type ‘void *’ used in arithmetic
rcdf.c: In function ‘RCDFReadR’:
rcdf.c:332: warning: pointer of type ‘void *’ used in arithmetic
rcdf.c: At top level:
/home/aburrell/bin/cdf38_0-dist/include/cdf.h:81: warning: ‘TT2000NULL’ defined but not used

and none of those are errors, yay!  However, I can't find the correct place to add the `-std=gnu99` flag so that the makefiles and make scripts will use it.  I tried altering $RSTPATH/build/make/makebin.linux by adding `-std=gnu99` to the end of the first line of code.  Although this seems like it should work, it doesn't?

This is currently preventing me from using RST.  There are issues with the newer computer I was using, so we were booted back to an older system.  This system has previously successfully installed RST.

## Priority

- [ ] Minor - typo, incorrect naming, fails in a specific uncommon situation
- [ ] Major - misrepresentation of data, failure in a common situation
- [x] Urgent - failure in usage, large misrepresentation of data

## RST version

*Does the bug occur in an official RST release?*

- [ ] Yes. RST version: ______
- [x] No. Branch containing the bug (e.g. develop): make_fov

## Example of the bug

See first section

### Attempts

See first section

### Data Access

N/A

## Potential Bug Location

$RSTPATH/build/make/makebin.linux

## Potential Solution(s)

See first section

## Extra Notes

See first section

aburrell, 2022-03-01:
That works, thanks for pointing out the makelib.linux line!  Now I am running into trouble because the compiler isn't recognizing `complex.h` in the old AACGM routine 🐞   I haven't been able to track down anything helpful on StackOverflow/google.  Any experience with this?

```
Compiling Binary:/home/aburrell/Git/rst/codebase/analysis/src.bin/aacgm/aacgmeval.1.2
================================================================================
make clean
rm -f *.o
rm -f version.h
rm -f errstr.h
rm -f hlpstr.h
rm -f aacgmeval
make
make.version /home/aburrell/Git/rst/codebase/analysis/src.bin/aacgm/aacgmeval.1.2
make.help
./doc/aacgmeval.doc.xml
cc -fPIC -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX -std=gnu99 -D_SVGLIB_ -I/home/aburrell/Git/rst/include/base -I/home/aburrell/Git/rst/include/general -I/home/aburrell/Git/rst/include/analysis -c aacgm.c    
mkdir -p /home/aburrell/Git/rst/bin
cc -L/home/aburrell/Git/rst/lib -o /home/aburrell/Git/rst/bin/aacgmeval aacgm.o -Wl,-Bstatic \
                           -laacgm.1 -lopt.1 -lrtime.1 -laacgm_v2.1 -ligrf_v2.1 -lastalg.1  -Wl,-Bdynamic \
                           -lm   
/home/aburrell/Git/rst/lib/libaacgm.1.a(rylm.o): In function `rylm':
rylm.c:(.text+0x9a): undefined reference to `CMPLX'
rylm.c:(.text+0xb9): undefined reference to `CMPLX'
rylm.c:(.text+0x1bd): undefined reference to `CMPLX'
rylm.c:(.text+0x1ea): undefined reference to `CMPLX'
collect2: ld returned 1 exit status
make: *** [aacgmeval] Error 1
Compilation Aborted.
```

aburrell, 2022-03-01:
> might as well apply on the whole make build system

That's my thought.  Could be done outside of the make_fov branch to keep the PR clean and well scoped.

mts299, 2022-03-01:
Sounds good but I am not assigning myself to that PR :p I have other things to do :) but should be simple for anyone todo minus whatever the fix is for complex.h

aburrell, 2022-03-01:
Really hopeful someone else has encountered the complex.h error

aburrell, 2025-02-14:
Have confirmed that the fix works.

closed 2025-02-14 by aburrell