SuperDARN/rst #486: map_addimf -wind xxx  later, map_plot can not find imf clock

J-LeeLC, 2022-03-20:
# BUG

*description of the bug*

## Priority

- [ ] Minor - typo, incorrect naming, fails in a specific uncommon situation
- [ ] Major - misrepresentation of data, failure in a common situation
- [ ] Urgent - failure in usage, large misrepresentation of data

## RST version

*Does the bug occur in an official RST release?*

- [ ] Yes. RST version: ______
- [ ] No. Branch containing the bug (e.g. develop): ______

## Example of the bug

*please provide **code** and **output** of the bug*

### Attempts

*What have you tried already to fix it?*

*What have you tried to work around it?*

### Data Access

*Please describe how to obtain the data needed to reproduce the bug* (e.g. specific file, use `make_fit` on any rawacf file).

## Potential Bug Location

*If possible, provide a link to the RST source code where you think the bug is coming from*

## Potential Solution(s)

*If possible, provide suggestions on how to fix the bug*

## Extra Notes

*Please provide any other details about this bug*

- *Operating System*
- *Compiler version* `gcc --version`
- *Examples of what the visual should look like*
- *Other data files that show the same bug*
- *How to access data files to test out the bug?*

ecbland, 2022-03-29:
@J-LeeLC Could you please provide an example of the bug? That will help us to investigate it. Thanks

ecbland, 2022-04-27:
@J-LeeLC I haven't heard back from you so I will close this issue. Please reopen it if you need further assistance.

closed 2022-04-27 by ecbland