SuperDARN/rst #490: Enabling separate tdiff values for non-stereo two-channel data

pasha-ponomarenko, 2022-03-23:
# Discussion

**Topic:**
Enabling separate tdiff values for non-stereo two-channel data
## 

The new hardware file format intorduced in https://github.com/SuperDARN/rst/pull/455 contains two different values of hardware time delay, _tdiff,_ which allows to calibrate two different channels of the stereo mode data within a single run of the fitting code, _make_fit._ This feature represent a significant improvement as compared to the original version of the hardware files. 
It relies on the fact that the stereo mode data are recorded into two separate channels, 1 and 2, while the "mono" data are generally recorded into channel 0. The current implementation utilises the _offset_ field in the parameter structure to distinguish between the stereo and "mono" modes so that all non-stereo data utilise only a single value of _tdiff_ regardless of the channel number. This pre-selection is applied to both FITACF2.5  https://github.com/SuperDARN/rst/blob/db5204d4f049e2e84817b6f89752fea0ad79c659/codebase/superdarn/src.lib/tk/fitacf_v3.0/src/fitacftoplevel.c#L174 
and FITACF3.0
https://github.com/SuperDARN/rst/blob/db5204d4f049e2e84817b6f89752fea0ad79c659/codebase/superdarn/src.lib/tk/fitacf.2.5/src/fitacf.c#L69
While the mono data are usully recorded with zero offset and channel = 0, **there is a significant volume of non-stereo data recorded with zero offset but split into channels 1 and 2** in order to distinguish between two radar frequency bands. These are two-frequency data from SAS, PGR, INV, RKN and CLY from late 2016 until the radar hardware was replaced by new BOREALIS system during 2019-2021. 

**I suggest to extend the benefits of the two-channel calibration to these and any other data recorded into channels 1 and 2 by removing the zero offset condition and retaining data pre-selection by the channels only.**

Implementing these changes requires very simple modifications of the code, and their testing is straightforward. Suitable test data would be from CLY in 2017 when there were two distinct _tdiff_ values at two channels:

- channel 1 (10 MHz): -13.5 ns
- channel 2 (12 MHz): -17.0 ns

egthomas, 2022-03-25:
@pasha-ponomarenko thank you for elaborating on your proposed changes.  I can see the value added by enabling multiple tdiff values to support the dual-frequency operation, however I do have a few questions and concerns (which are my own thoughts and not on the behalf of the other Tdiff task force members):

1. Have the same frequency bands always been used on each channel at each radar?  For example, throughout CLY's history, has the 10 MHz band always been assigned to channel 1 and the 12 MHz band to channel 2?  For more recent interleavesound and normalsound campaigns, I believe different frequencies may have been assigned to channels 1 and 2 (and 3,4,5, etc) which may not be the same as the twofsound bands.
2. Similarly, how will the proposed changes affect the processing of non-twofsound data, when the radars may (or may not) alternate between day and night frequencies?  Won't the twofsound channel 1 tdiff value then be applied to both the daytime and nighttime frequency bands?  (I'll concede here that using a single fixed value for day/night operations isn't necessarily appropriate either)
3. How will the proposed change affect other non-Stereo radars that do have multiple ("physical") channels?  For example, the UAF / Penn State-style radars have up to 4 channels which may or may not require separate tdiff values - if KOD is typically running on channel 4 (d), then won't the assigned `tdiff` value always come from channel 2 in the `hdw` file, even if the (default) channel 1 value should be used?

I guess my main concern is that I see this as a short-term solution to a problem the task force plans to address more comprehensively, and I worry about some of the unintended consequences when the suggested modifications are applied more broadly.

pasha-ponomarenko, 2022-03-25:
@egthomas , thanks a lot for your points!

> Have the same frequency bands always been used on each channel at each radar? For example, throughout CLY's history, has the 10 MHz band always been assigned to channel 1 and the 12 MHz band to [...]

For `twofsound`, the answer would be "yes". Our `normalsound` regimes normally use the lower frequency  band of the twof mode, so this should not be a problem either.  

On `interleavesound`, I believe it is not used in the common time, and on a more general note we cannot cater for all possible  cases anyways.  With that respect, different frequencies can be also assigned at different times to the same channels for `stereo` radars, so the whole approach with channel-assigned `tdiffs` is prone to that.  

> Similarly, how will the proposed changes affect the processing of non-twofsound data, when the radars may (or may not) alternate between day and night frequencies? Won't the twofsound channel 1 [...]

I also don't think this is an issue because a single `tdiff` value is used here anyways.

> How will the proposed change affect other non-Stereo radars that do have multiple ("physical") channels? For example, the UAF / Penn State-style radars have up to 4 channels which may or may not [...]

I though about this before, and this issue can be easily addressed by assigning the same value to both `tdiffs`. Furthermore, it will be advantageous for such radars as it would allow in principle to calibrate two channels instead of only one. So, I don't see any problems here either.

In summary, I believe we should cater first of all for the majority of the commonly used data, and the changes proposed in this issue are achieving this goal by extending proper claibration onto a large amount of the common mode data.

pasha-ponomarenko, 2022-03-31:
I double-checked the "lettered" files from UAF radars, and all of them use channel 0 anyways:
```
ponomarenko@pasha-linux-main:~/data/fit> dmapdump 20211225.1000.00.kod.d.rawacf | grep 'channel' | head -5
        short   "channel" = 0
        short   "channel" = 0
        short   "channel" = 0
        short   "channel" = 0
        short   "channel" = 0
```

closed 2022-12-21 by egthomas