SuperDARN/rst #522: Calling the `lmfit2` library from `make_fit` -- duplicated functions with `fitacf3`

ecbland, 2022-08-19:
I'm looking for advice on how to implement the final step of our updates to `make_fit`: adding a `-lmfit2` command line option and removing the `make_lmfit2` binary. The challenge here is that `fitacf_v3` and `lmfit_v2` contain a lot of functions with the same names (and sometimes there are some minor differences between them), and of course we have to include both libraries when compiling `make_fit`.

Some options are, in order of the amount of effort required...

- Keep the `make_lmfit2` binary, but add a note to the `make_fit` command line help about how to run `lmfit2`. 
- Rename all of the functions in `lmfit2` so that all function names are unique (e.g. add a prefix "lmfit2_" to everything)
- Create a separate library for the functions that are used by both `fitacf_v3` and `lmfit_v2` so that there is only one version in the codebase. The ACF pre-processing functions are a good example here. Where this is not possible, rename the functions as above.
- **Any other options?**

pasha-ponomarenko, 2022-08-19:
@ecbland , I would go with the most economic way as it is hard to imagine an extensive usage of this package until the processing speed is drastically improved.

egthomas, 2022-08-19:
I tend to agree with @pasha-ponomarenko.

@ecbland similar to your first suggestion, could we also add a `-lmfit2` option to `make_fit` that just prints a message telling the user to call `make_lmfit2` instead and then exits?  Sometimes the command line help can become too long and users might miss that detail.

closed 2022-08-24 by ecbland