SuperDARN/rst #529: Plan for RST 5.0 release

ecbland, 2022-10-13:
We are due for a new release of the RST! Lots of new features and functionality have been added in the last few months, so the next version of the RST will be a major release, RST5.0 🎉 

We are still waiting for #512 and #509 to be merged. **If you're working on anything else that should be included in the next release (and will be easy to test), please leave a comment below.**

The deadline for PI feedback on #512 is 18th October, so I anticipate that we could create the release branch at the end of next week, e.g. 21st October.

egthomas, 2022-12-21:
@ecbland thanks for all your hard work on the 5.0 release and as DAWG chair over the last few years!

I've just updated the `gh-pages` branch with the new XML / HTML documentation and added the 5.0 release to the `rst-archive` repository.  Think we're ready to close this issue?

pasha-ponomarenko, 2022-12-21:
@ecbland, 👍👏🎉😢

ecbland, 2022-12-21:
> @ecbland thanks for all your hard work on the 5.0 release and as DAWG chair over the last few years!
> 
> I've just updated the `gh-pages` branch with the new XML / HTML documentation and added the 5.0 release to the `rst-archive` repository. Think we're ready to close this issue?

Thanks for doing that @egthomas, and for the kind words 😄   Yes, this issue can be closed now 💯

closed 2022-12-21 by ecbland