SuperDARN/rst #548: make_grid / Chisham VHT model bug on Tiger Bay

alexchartier, 2022-12-02:
# BUG

I get a weird error when running make_grid with -chisham option on Tiger Bay data. The issue does not come up with other radars. 
I'm not sure if it's indicative of a wider problem. 

## Priority

- [X] Minor - typo, incorrect naming, fails in a specific uncommon situation
- [ ] Major - misrepresentation of data, failure in a common situation
- [ ] Urgent - failure in usage, large misrepresentation of data

## RST version

*Does the bug occur in an official RST release?*

- [ X] Yes. RST version: 4.4.1

## Example of the bug

e.g. make_grid -chisham 20150316.tig.v3.0.fit 20150316.tig.v3.0.grid

ERROR: coefficients are not valid for altitudes above 2000 km: 3618.313121.
       You must either use field-line tracing (TRACE or ALLOWTRACE) or
       indicate that you know this is a very bad idea (BADIDEA)
[multiple errors like this]

### Attempts
Occurs on multiple files, only TIG with -chisham

ecbland, 2022-12-02:
Thanks @alexchartier for the bug report. You're using an old version of RST, but I've confirmed this also occurs on the RST5.0 release branch for the data file in your example. 

However, that particular data you are using is special. Starting from 10:00 UT, the radar was running `longrangescan`, which was a fun discretionary-time experiment I did during my PhD to find out if the radar could detect multi-hop groundscatter from ~10,000km away. I ran the same experiment at Buckland Park in December 2014. The geolocation and AACGM calculations are simply not going to be valid for such long ranges, so the software rightly reports an error.

So I'd say this is not a bug, but I recommend you exclude that data from your grid file.

alexchartier, 2022-12-03:
Thanks Emma - I agree that it's not a bug in that case. The only caveat would be that if make_grid returns values after giving the error then I'd say that's a bug. The error message seems to indicate that I'd have to provide a "-BADIDEA" flag to allow those data points to be returned. I would also suggest adding "MAKE_GRID" to the start of the error message. It took me some time to work out where this was coming from

egthomas, 2022-12-05:
@alexchartier note there are `-maxrng` and `-maxsrng` options available for `make_grid` which can be used to specify the maximum range gate number or slant range, respectively.  There are a few pros and cons to each, eg `-maxrng` is faster but won't work well with non-standard range resolution modes (< 45 km), while `-maxsrng` can handle any range resolution but is slower because of the additional gate-to-range calculations.

alexchartier, 2022-12-05:
Thanks @egthomas  - what would you say is the longest distance you'd trust for F-region convection mapping? 3000 km?

alexchartier, 2022-12-05:
So the plot thickens a bit - I thought this command would eliminate the coefficient error, but it did not:
make_grid -xtd -chisham -ion -maxsrng 10   20150316.tig.v3.0.fit

I tried with various other values of maxsrng to no avail. Also, I don't understand why these TIG files are the only ones to trigger the error, or exactly what it means. The error is "coefficients are not valid for altitudes above 2000 km". I don't see how >2000 km **altitudes** would be relevant in this (or any) case. On the other hand, >2000 km **ranges** would be relevant to many (all?) radars

egthomas, 2022-12-05:
@alexchartier a command line argument of `-maxsrng 10` means maximum slant range of 10 km, which is probably not what you want.  3000 km is probably a safe cut-off, although really that's best left to the user's discretion depending on the application (eg instantaneous vs statistical patterns).

The reason this error happens is that the Chisham et al. model finds a virtual height which increases non-linearly with range for (assumed) 1.5-hop ionospheric scatter observed beyond about 2100 km slant range.  At the incredibly long ranges of Emma's special mode, this means the resulting virtual heights are likely well above the altitudes for which AACGM coordinate transformations (using the v2 coefficients) are valid, hence the AACGM-specific error message.

egthomas, 2022-12-05:
@alexchartier ok no problem, just wanted to make sure.  You are correct that the code does appear to map all of the data first and then determine whether to reject the range / beam cell based on the various filters.

I've just pushed a possible fix to the `grid_range_fix` branch - let me know if this solves the problem: https://github.com/SuperDARN/rst/commit/afbb7b0cee2bc930cd63d8f323c60184e1a3d9c8

alexchartier, 2022-12-05:
Looks good - thanks!

closed 2022-12-05 by alexchartier

egthomas, 2022-12-05:
@alexchartier after looking at this a little bit more closely, I'm not sure if my fix was a comprehensive one.

`GridTableAddBeam` (where the geolocation happens) is supposed to be called once for each azimuthal beam number of a radar scan.  My fix was to check the `sct` flag to determine whether to skip a given range/beam cell in `GridTableAddBeam`.  But the `sct` flag for a given beam/range cell might not always be set to `0` solely because of the range (or other filtering) - it may have been set to `0` because there were no data on a given beam sounding.

So if data were missing on the first beam of a scan, the range/beam positions may not be correctly populated for subsequent beam soundings in the scan.