SuperDARN/rst #563: install RST - Centos 8

Aixuwu, 2023-04-04:
Excuse me, ask a question, See from Table of Contents-Library Requirements for installing RST, CentOs only has two versions of 6.1 and 7. How to install RST on CentOs 8?

Aixuwu, 2023-04-04:
I try to install as CentOs 7
sudo yum install hdf5-devel libpng12-devel libX11-devel libXext libXext-devel netcdf netcdf-devel ncurses-devel zlib-devel,
then error: no match: libpng12-devel

egthomas, 2023-04-04:
@Aixuwu can you try installing `libpng-devel` instead of `libpng12-devel` ?

ecbland, 2023-04-05:
I recently installed RST on CentOS 7 (and I was planning to share my notes...just juggling lots of things at the moment in my new job). 

I needed to do the following to get RST to compile:
1. Install `libpng-devel` instead of `libpng12-devel` as  @egthomas suggested above
2. Add `-std=gnu99` to line 10 of `$RSTPATH/build/make/makebin.linux` and line 10 of `$RSTPATH/build/make/makelib.linux`
   e.g. `CFLAGS+=-fPIC -std=gnu99 -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX`
3. I could not get `fitacfclientgui` to compile, so as a workaround I compressed the directory `$RSTPATH/codebase/superdarn/src.bin/tk/tcpip/fitacfclientgui.1.0` so that RST will skip over it. We had a similar problem with macOS (#536). It would be good to find a proper solution.

egthomas, 2023-04-05:
Thanks @ecbland! For `fitacfclientgui`, you may also need to install `ncursesw5-devel` (or whatever the equivalent is on CentOS) - that was required on another recent installation I tried on an older Ubuntu version.