SuperDARN/rst #572: WAL Coordinates

JWiker, 2023-05-31:
# BUG

The coordinates for WAL or slightly off - they list Wallops proper instead of the exactly location of the array

JWiker, 2023-05-31:
@egthomas Would you be able to grant me access/permissions so I can push to this issue's branch?

egthomas, 2023-06-01:
@JWiker it appears you're already a member of the rst developers team so you should have write access to this repository.  What kind of error message are you encountering?

ecbland, 2023-06-01:
@JWiker If this is an update to the WAL hardware file then those changes should be made in the [`hdw` repo](https://github.com/SuperDARN/hdw), and then we will copy them over to RST for the next release. 

You should have write access to the `hdw` repo, although perhaps you will get the same error messages that you're getting here 😟 In any case we should sort that out.

egthomas, 2023-06-01:
For context, I believe this came out of a higher-level discussion at this week's SuperDARN workshop about updates to all of the radar latitude/longitude locations in the hdw.dat files.  Basically, we were able to access commercial-grade satellite imagery through one of our NSF-sponsored projects to achieve much better accuracy than our previous attempt in #466 (which used Google Earth).  Hopefully something will filter down from the PIs soon so we can share everything at once.

JWiker, 2023-06-06:
> @JWiker If this is an update to the WAL hardware file then those changes should be made in the [`hdw` repo](https://github.com/SuperDARN/hdw), and then we will copy them over to RST for the next [...]

Thanks @ecbland, I'll make the changes there

JWiker, 2023-06-06:
> Thanks @ecbland, I'll make the changes there

A ha, looks like Evan has already done it. Thanks!

closed 2023-06-06 by JWiker