SuperDARN/rst #584: adding IMF file

ArdraRamachandran, 2023-07-29:
**Issue: Unable to add IMF data when creating a MAP file from GRID file in SuperDARN RST 5.0**

**Description:**
I have previously installed SuperDARN RST 5.0 on my Ubuntu machine, and I am encountering an issue while attempting to create a MAP file from a GRID file. The problem arises when I try to add the IMF (Interplanetary Magnetic Field) data using the following command:

map_addimf -if [IMF ASCII file path] -d 00:00 $year$month$day.hmb.map > $year$month$day.imf.map

While the `map_addimf` command runs without displaying any errors, the resulting `$year$month$day.imf.map` file does not contain the expected IMF data. The generated map file appears to be incomplete, lacking the necessary information from the IMF file.

**Steps to Reproduce:**
1. Installed SuperDARN RST 5.0 on Ubuntu.
2. Prepared a GRID file in the required format.
3. Ensured the IMF ASCII file adheres to the correct format as per the SuperDARN RST documentation.
4. Executed the `map_addimf` command with appropriate options and file paths.

**Expected Behavior:**
The `map_addimf` command should successfully add the IMF data to the MAP file, generating the `$year$month$day.imf.map` file containing the appropriate information.

**Actual Behavior:**
The `$year$month$day.imf.map` file is generated, but it does not include the expected IMF data. The command completes without showing any error messages or indications of the issue's cause.

**Additional Information:**
- Operating System: Ubuntu 22.04 LTS
- SuperDARN RST Version: 5.0

**Note:**
The IMF ASCII file used during testing has been thoroughly validated to ensure its adherence to the format specified in the SuperDARN RST documentation.

Please let me know if there are any additional logs, information, or troubleshooting steps required to help diagnose and resolve this issue. I appreciate your assistance. Thank you!

egthomas, 2023-07-31:
@ArdraRamachandran when you say the output map file does not contain the expected IMF information, do you mean that 1) no IMF information is being added, or 2) the IMF information is being added to the wrong time record, or 3) entirely different IMF information is being added, or 4) something else?

I tried replicating this issue but was unable to do so with the attached IMF text file (where the IMF By component varies at each time) and following these steps:

```
map_grd -empty -sd 20230101 -st 00:00 -ex 00:30 -tlen 120 > 20230101.empty.cnvmap
map_addhmb -nodef 20230101.empty.cnvmap > 20230101.hmb.cnvmap
map_addimf -if imf.txt -d 00:00 20230101.hmb.cnvmap > 20230101.imf.cnvmap

dmapdump 20230101.imf.cnvmap > 20230101.imf.cnvmap.dump
more 20230101.imf.cnvmap.dump
```

[imf.txt](https://github.com/SuperDARN/rst/files/12217049/imf.txt)

ArdraRamachandran, 2023-08-01:
The data I was trying to analyze was the SuperDARN data on 22/06/2015. I generated a 300 seconds GRID file using the following commands:

radarlist="$(ls | cut -d "." -f 4 | sort -u)"
for radar in $radarlist; do make_grid -tl 300 -xtd -vb -ion -minrng 20 20150622*.$radar*fitacf > 20150622.$radar.grd; done
combine_grid -vb 20150622*.grd > 20150622.north.grd

Afterward, I proceeded with mapping the GRID data:

map_grd -empty -sd 20150622.north.grd -st 00:00 -ex 00:30 -tl 300 > 20150622.empty.map
map_addhmb 20150622.empty.map > 20150622.hmb.map
map_addimf -vb -if June_IMF.txt -d 00:00 20150622.hmb.map > 20150622.imf.map

The resulting output is as follows:
2015-6-22 0:0:0 delay=0:0 Bx=0 By=0 Bz=0 Vx=0 Kp=0
2015-6-22 0:5:0 delay=0:0 Bx=0 By=0 Bz=0 Vx=0 Kp=0
2015-6-22 0:10:0 delay=0:0 Bx=0 By=0 Bz=0 Vx=0 Kp=0
2015-6-22 0:15:0 delay=0:0 Bx=0 By=0 Bz=0 Vx=0 Kp=0
2015-6-22 0:20:0 delay=0:0 Bx=0 By=0 Bz=0 Vx=0 Kp=0
2015-6-22 0:25:0 delay=0:0 Bx=0 By=0 Bz=0 Vx=0 Kp=0

For reference, you can find the visual representation of the data in the figure linked below (you can see the IMF data is missing):
[image]

Additionally, I used the attached "June_IMF.txt" file to incorporate interplanetary magnetic field (IMF) data into the mapping process.

If you need any further information please let me know,
[June_IMF.txt](https://github.com/SuperDARN/rst/files/12224157/June_IMF.txt)

PS: I tried manually adding the data in the format;
map_addimf -bx 1.5 -by -1.2 -bz 0.4 [file].map > [file].imf.map 
which worked well.

ArdraRamachandran, 2023-08-02:
Yes, it is working. Never tried that one before, Thank you!

closed 2023-08-02 by egthomas