SuperDARN/rst #603: Problem help

wonenye, 2023-10-19:
Why do I appear after compiling code with make.code
Failed to locate and error idl_export.h occurred. I used the method on the document but it didn't work, could you help me? Thank you

egthomas, 2023-10-19:
@wonenye do you have IDL / are you trying to install the IDL portion of the RST, or are you trying to skip over it?

wonenye, 2023-10-20:
I found the idl.bash file in the rst/.profile/idl.bash directory and it says
# idl.bash
# ========

# IDL header directory

export IDL_IPATH="/usr/local/itt/idl/external/include"

# RST DLM directory

export DLMPATH="${RSTPATH}/dlm"

# IDL environment

export IDL_PATH="<IDL_DEFAULT>:+/${RSTPATH}/idl/lib"
export IDL_DLM_PATH="<IDL_DEFAULT>:${RSTPATH}/dlm"
export IDL_STARTUP="${RSTPATH}/idl/startup.pro"

# Makefile prototype for IDL DLMs

export MAKEDLM=${BUILD}/make/makedlm
, I don't know if this proves that I have IDL?

wonenye, 2023-10-20:
I found the idl.bash file in the rst/.profile/idl.bash directory and it says,
# idl.bash
# ========
# IDL header directory
export IDL_IPATH="/usr/local/itt/idl/external/include"
# RST DLM directory
export DLMPATH="${RSTPATH}/dlm"
# IDL environment
export IDL_PATH="<IDL_DEFAULT>:+/${RSTPATH}/idl/lib"
export IDL_DLM_PATH="<IDL_DEFAULT>:${RSTPATH}/dlm"
export IDL_STARTUP="${RSTPATH}/idl/startup.pro"
# Makefile prototype for IDL DLMs
export MAKEDLM=${BUILD}/make/makedlm
 I don't know if this proves that I have IDL?

wonenye, 2023-10-20:
apart from that，when l install yum -y  install libhdf5-serial-dev libncurse-dev libnetcdf-dev libpng-dev libx11-dev libxext-dev， Terminal display “nopackage available”，Could you help me with that? Thank you