SuperDARN/rst #604: Installation issue

alok342311, 2023-11-17:
# Discussion

**Having initial problem with installation**

## I want to use pydarn in my system for polar convection maps, for that I have to convert FITACF to GRID to MAP files... I am trying to install DAWG rst software for that, however I am getting the following error during make.build (I am using the instructions provided on https://radar-software-toolkit-rst.readthedocs.io/en/latest/user_guide/linux_install/#installation)

(base) space@Space:~/rst/build/script$ make.build
rm -f *.o
make.version /home/space/rst/codebase/base/src.lib/xml/xml.1.8/src/..
make.hdr /home/space/rst/codebase/base/src.lib/xml/xml.1.8/src/../include /home/space/rst/build/include/base 
/home/space/rst/codebase/base/src.lib/xml/xml.1.8/src/../include/rxml.h
'/home/space/rst/codebase/base/src.lib/xml/xml.1.8/src/../include/rxml.h' -> '/home/space/rst/build/include/base/rxml.h'
/home/space/anaconda3/bin/x86_64-conda_cos6-linux-gnu-cc -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/space/anaconda3/include -fPIC -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX -I/home/space/rst/build/include/base -c rxml.c    
/home/space/anaconda3/bin/x86_64-conda_cos6-linux-gnu-cc -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/space/anaconda3/include -fPIC -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX -I/home/space/rst/build/include/base -c call.c    
mkdir -p /home/space/rst/build/lib
ar -r /home/space/rst/build/lib/librxml.1.8.a rxml.o call.o
ar: creating /home/space/rst/build/lib/librxml.1.8.a
/home/space/anaconda3/bin/x86_64-conda_cos6-linux-gnu-ld -shared -lm -lz  -o /home/space/rst/build/lib/librxml.1.8.so \
              rxml.o call.o -Bstatic -L/home/space/rst/build/lib  
/home/space/anaconda3/bin/x86_64-conda_cos6-linux-gnu-ld: cannot find -lz
make: *** [/home/space/rst/build/make/makelib.linux:25: rxml] Error 1

alok342311, 2023-11-17:
please help

JWiker, 2023-11-17:
Hi @alok342311  - have you confirmed that zlib is installed? If it is, is it also part of your LD_LIBRARY_PATH?

alok342311, 2023-11-17:
Hello @JWiker .... thanks for your reply... the following massage I am getting for zlib1g ... I do not use linux frequently, I appreciate your guidance..... which path should I export for  LD_LIBRARY_PATH

(base) space@Space:~$ dpkg -L zlib1g
/.
/lib
/lib/x86_64-linux-gnu
/lib/x86_64-linux-gnu/libz.so.1.2.11
/(base) space@Space:~$ dpkg -L zlib1g
/.
/lib
/lib/x86_64-linux-gnu
/lib/x86_64-linux-gnu/libz.so.1.2.11
/usr
/usr/share
/usr/share/doc
/usr/share/doc/zlib1g
/usr/share/doc/zlib1g/changelog.Debian.gz
/usr/share/doc/zlib1g/copyright
/lib/x86_64-linux-gnu/libz.so.1
usr
/usr/share
/usr/share/doc
/usr/share/doc/zlib1g
/usr/share/doc/zlib1g/changelog.Debian.gz
/usr/share/doc/zlib1g/copyright
/lib/x86_64-linux-gnu/libz.so.1

JWiker, 2023-11-17:
Try `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/lib/x86_64-linux-gnu` - I think that should work since that's where the actual shared library files are located

alok342311, 2023-11-17:
It is showing the same error....

(base) space@Space:~/rst/build/script$ export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/lib/x86_64-linux-gnu
(base) space@Space:~/rst/build/script$ make.build
/bin/bash: /home/space/anaconda3/lib/libtinfo.so.6: no version information available (required by /bin/bash)
rm -f *.o
make.version /home/space/rst/codebase/base/src.lib/xml/xml.1.8/src/..
make.hdr /home/space/rst/codebase/base/src.lib/xml/xml.1.8/src/../include /home/space/rst/build/include/base 
/home/space/rst/codebase/base/src.lib/xml/xml.1.8/src/../include/rxml.h
'/home/space/rst/codebase/base/src.lib/xml/xml.1.8/src/../include/rxml.h' -> '/home/space/rst/build/include/base/rxml.h'
/home/space/anaconda3/bin/x86_64-conda_cos6-linux-gnu-cc -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/space/anaconda3/include -fPIC -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX -I/home/space/rst/build/include/base -c rxml.c    
/home/space/anaconda3/bin/x86_64-conda_cos6-linux-gnu-cc -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/space/anaconda3/include -fPIC -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX -I/home/space/rst/build/include/base -c call.c    
mkdir -p /home/space/rst/build/lib
ar -r /home/space/rst/build/lib/librxml.1.8.a rxml.o call.o
ar: creating /home/space/rst/build/lib/librxml.1.8.a
/home/space/anaconda3/bin/x86_64-conda_cos6-linux-gnu-ld -shared -lm -lz  -o /home/space/rst/build/lib/librxml.1.8.so \
              rxml.o call.o -Bstatic -L/home/space/rst/build/lib  
/home/space/anaconda3/bin/x86_64-conda_cos6-linux-gnu-ld: cannot find -lz
make: *** [/home/space/rst/build/make/makelib.linux:25: rxml] Error 1