SuperDARN/rst #613: Cross-interference in HOK/HKW data

pasha-ponomarenko, 2024-04-24:
# Discussion

**Topic: Mitigating cross-interference effects between HOK and HKW radars**

## Description

Since the inception of HKW radar in 2014 both HOK and HKW datasets appear to contain an excessive amount of "salt-and-pepper" noise. Importantly, it disappears in one radar when another one is not operating. A detailed investigation during my stay in Japan in early 2023 revealed that this interference is caused by power spikes surrounding additional blanking pulses designed to protect receiving circuitry in one radar from emission of the other one. As the radars use different pulse sequences and number of range gates, the additional blanking pulses are distributed more or less randomly across all range gates. This component is highly visible in FITACF3 data but it is effectively "swept under the carpet" in FITAF2.5 data by overfiltering together with a large amount of otherwise valid data. The HOK/HKW SuperDARN engineer Mr Hamaguchi suggested that we can effectively remove the switch-related power spikes in I&Q data by identifying the pulse locations through low values of sample power. Our preliminary tests using IDL DLLs form RST showed that this approach works for at least some typical data, and now we should implement the I&Q filtering in RST. Emma has already started working on this.

A detailed description of this artifact and preliminary results of its mitigation can be found in our SD'23 presentation "Cross-interference effects at Hokkaido East/Hokkaido West radars: source identification and problem mitigation"
([2-Ponomarenko_Interference.pdf](https://github.com/SuperDARN/rst/files/15097360/2-Ponomarenko_Interference.pdf))
 

## Category

- [ ] User-friendly
- [X] Software Design
- [X] Data related
- [ ] Capabilities
- [ ] Clarity
- [ ] Workflow

## Details

*Please provide any other details or description around your question.*

## Example

Example of cross-interference:

[image]

Data without correction:

[image]

Data with correction:

[image]

egthomas, 2024-04-24:
@pasha-ponomarenko has there been any change in the interference behavior after HKW increased the number of range gates (starting in ~April 2023)?  And do you know if there is a similar issue at other co-located radar sites such as Syowa or Dome C?

pasha-ponomarenko, 2024-04-24:
@egthomas, it looks like HKW switched to 110 gate on 28 March 2023. It did not change in any noticeable way the noise/interference background at HKW, which remained quite bad. It also seems like the overall noise/interference situation at HKW became worse on 2 April 2022. 
I am not sure about Syowa or Dome C. I would expect that these pairs are synchronised, which is easier to do if both equipment sets are located in the same place. This is not the case with HOK/HKW. They have a length of fiber line laid between the two sites, but it is only used to trigger local blanking pulses when the other radar is emitting. This leads to the clutter from their local T/R switches to be distributed randomly across range gates.

pasha-ponomarenko, 2024-06-17:
Just adding a comment from Sessai on 24 April 2024:
"Dear Pasha,
I'm not sure/confident how I can respond to messages on github, so let me admit to do so with usual email.
sys/sye pair is not synchronized (as hop radar pair) the 2 radars are relatively close but apart several hundred  meters apart and not located at the same place (like many of newer mid-lat. radars).
hop (and possibly DomeC) use(s) the same idea with sys/sye ways, i.e., 2 (un-synchronised) BASBOXes (apart each other) sends blanking signals when own radar transmits pulses to the other radar('s basbox) each other which sets "ALL_ATTEN=on" when receiving Blanking is on, and the other radar do not know the timing of reception of blanking signal, and as Pasha said, the attenuated Rx signals produce new clutters.
This way of sending blanking signals each other starts when sye started operation in 1997. During the preparation phase before that, Mike Pinnock at BAS and Hisao Yamagishi @ NIPR tried to think about the way to avoid damage of receivers at twin radars. Nozomu and probably Gigi Coco?
asked me about the way and at least Nozomu adopted the way.
Cheers,
Sessai at Syowa
"

pasha-ponomarenko, 2024-06-17:
So it looks like this issue might have a bit wider scope covering Syowa and Dome C radars as well. Now, I wonder if any of these radars keep I&Q data.