SuperDARN/rst #626: Installation Issue - Compiling Library:/home/linux/RST/rst//codebase/analysis/src.lib/cdf/rcdf.1.5

mj299, 2024-10-17:
Linux as well as RST newbie here - so please be gentle!

I'm trying to install RST onto OpenSUSE Leap 15.6. I'm attempting to follow the installation instructions online - [here](https://radar-software-toolkit-rst.readthedocs.io/en/latest/user_guide/linux_install/#cdf-library).  

I believe the library requirements installed OK. As far as I can tell the command worked OK - 
`sudo zypper install gcc hdf5-devel libpng16-devel libX11-devel libXext6 libXext-devel netcdf netcdf-devel ncurses-devel zlib-devel`.

I unpacked and installed the CDF library. This also seemed to go OK.

The installation of RST is where I have come unstuck. RST is in the following path:
`/home/linux/RST/rst/`

.profile.bash contains the following lines
`export OSTYPE="linux"
export SYSTEM="linux"`

I modified the path in 'rst/.profile/base.bash' to point where cdf.h is located:
`export CDF_PATH="/usr/local/cdf/include"`
If I navigate to this path on the command line, I can see that cdf.h is there.

I have run `source ~/.bashrc` to reload the environment variables. 

I then changed the directory to `$RSTPATH/build/script` and ran `make.build`. This appears to work OK. `make.code` ends in a 'compilation aborted'. The the 'Compiling Library:/home/linux/RST/rst//codebase/analysis/src.lib/cdf/rcdf.1.5' stage. It appears to relate to the location of cdf.h:
```
rcdf.c:36:10: fatal error: cdf.h: No such file or directory
 #include "cdf.h"
          ^~~~~~~
compilation terminated.
```
Here is the full log: [superdarn-rst.build.20241017.1510.zip](https://github.com/user-attachments/files/17413940/superdarn-rst.build.20241017.1510.zip)

Any help greatly appreciated!

egthomas, 2024-10-18:
@mj299 it looks like you may just need to trim the extra `include` you added to your `CDF_PATH` in `base.bash`.  For example, in your attached log file when compiling `rcdf.1.5` it shows

```
cc -fPIC -Wall -pedantic -O3 -D_GNU_SOURCE -D_LINUX -I/home/linux/RST/rst//include/analysis -I/usr/local/cdf/include/include -c rcdf.c
```

Note the `/usr/local/cdf/include/include` path being used. This is because the makefile for `rcdf` (and other libraries/binaries using CDF) appends the `include` subdirectory for you:

```
INCLUDE=-I$(IPATH)/analysis -I$(CDF_PATH)/include
```

Hope that helps!

mj299, 2024-10-18:
@egthomas - That worked! It all compiled and I've checked I can convert file types and produce a few plots with some sample data.

Many thanks!

closed 2024-10-18 by mj299