SuperDARN/rst #636: Return errors when calling map_fit

JouleCai, 2025-05-15:
# BUG

I got the error messages when calling `map_fit`:

2025-3-26 23:44:0 dp=24.4426 error=0.0310599 chi_sqr=1.52919e+08 rms_err=50638.1
ERROR: latitude must be in the range -90 to +90 degrees: 90.500000
2025-3-26 23:46:0 dp=24.6801 error=0.0310586 chi_sqr=1.54878e+08 rms_err=50181.2
ERROR: latitude must be in the range -90 to +90 degrees: 90.500000
2025-3-26 23:48:0 dp=26.7818 error=0.0242918 chi_sqr=1.6711e+08 rms_err=14617.7
ERROR: latitude must be in the range -90 to +90 degrees: 90.500000
2025-3-26 23:50:0 dp=29.1927 error=0.0235854 chi_sqr=1.82268e+08 rms_err=21328
ERROR: latitude must be in the range -90 to +90 degrees: 90.500000
2025-3-26 23:52:0 dp=30.8541 error=0.0234048 chi_sqr=1.94183e+08 rms_err=44289.1
ERROR: latitude must be in the range -90 to +90 degrees: 90.500000

No errors were found during the processes "RAWACF->FITACF" and "FITACF->GRID". 

## Priority

- [ ] Minor - typo, incorrect naming, fails in a specific uncommon situation
- [ ] Major - misrepresentation of data, failure in a common situation
- [ x] Urgent - failure in usage, large misrepresentation of data

## RST version

*Does the bug occur in an official RST release?*

- [ x] Yes. RST version: v5.1
- [ ] No. Branch containing the bug (e.g. develop): ______

## Example of the bug

*please provide **code** and **output** of the bug*
`map_fit -vb 20250326.north.ts18.model.map > 20250326.north.ts18.map`

JouleCai, 2025-05-19:
> [@JouleCai](https://github.com/JouleCai) thanks for raising this issue. Could you share the grid file you are working with so we could try troubleshooting the problem?

@egthomas Thank you for your reply. Please find the grid file and the map files in this folder:

https://drive.google.com/drive/folders/1I6TdJnIOF-eQDblXRlf5xGtkC2IZlGXD?usp=sharing

The options used for `make_grid`: '-vb', '-xtd', '-i 120',  '-tl 120', '-minsrng 800', '-maxsrng 2500', '-vmin 50', '-vmax 1500', '-vemax 500', '-ion', '-chisham'
These were applied to all the radars.

egthomas, 2025-05-19:
@JouleCai thank you.  I just downloaded and compiled a fresh installation of RST 5.1 and was not able to reproduce these MLAT > 90° results seen in your grid file (eg, in the first record for SAS (stid=5) and ICW (stid=210)).

Since all of the MLON values are -90° and all of the MLAT values are 90.5° in the shared file, I wonder if there is some issue with either the input FITACF files or the version of RST (or possible AACGM/IGRF) you are using?  Can you try executing `make_grid --version` and `echo $IGRF_COEFFS` and verifying the output?  Is this the only date for which you have encountered this issue?

JouleCai, 2025-05-20:
@egthomas The problem seems solved now. It's not related to the RST package, but associated with my environment's settings. 

I executed my Python script to batch multiple radar datasets in a Conda virtual environment. I used the package `subprocess` to call the RST shell commands. I also used `pydarn` to read the hardware information of the radars. The `aacgmv2` package was also installed in the environment. Now I create a new virtual environment with the latest versions of the dependencies. The problem is solved. I'm not sure if it's because the radar hardware information was read correctly with an older version of `pydarn`, or RST read the aacgmv2 coefficients from the aacgmv2 package in the conda environment instead of the default directory in the software folder.

I'm sorry for bothering you a lot. I appreciate your help very much.

aburrell, 2025-05-20:
It does sound like it was an evironmental issue.  The Python aacgmv2 package is set up to use your environment's AACGM coefficients by default, rather than those provided within the package, to help reduce problems with RST users or those doing something specific with the code.  So I think that is unlikely, but not impossible (depending on the order in which different packages were installed and invoked).