SuperDARN/rst #637: Possible holdover from old style hardware files

billetd, 2025-05-16:
I was converting `RadarLoadHardware` into python for a project I'm working on, and noticed at L213 of radar.c:

https://github.com/SuperDARN/rst/blob/434880f20fc170ff2fc1b97b0938938947892b9d/codebase/superdarn/src.lib/tk/radar.1.22/src/radar.c#L213

that it throws out files with less than 24 parameters on the line. The current format of hardware files have 22 parameters, so this line auto rejects a modern hardware file.

Just wanted to put this as an issue before making a PR to change it from 24 to 22. Does this matter anywhere else or is just a holdover from old hardware files?

egthomas, 2025-05-16:
@billetd I think the 22 vs 24 difference is because the `HH:MM:SS` column in the `hdw` file is treated by the `sscanf` line above as 3 separate parameters rather than one, so it seems like 24 is the correct number.

billetd, 2025-05-17:
@egthomas that makes sense, thanks, sorry I missed that. I think I thought it might be the case because I was using `strip()` in python to separate the whole line into chunks, which keeps `HH:MM:SS` as one. Probably saved myself an problem down the line posting this issue though!

closed 2025-05-17 by billetd